VoronDesign/Voron-2 #403: Cannot open Voron 2.4 step file

mhejjas, 2022-11-06:
When trying to import the Voron2.4_Assembly.step file into Solidworks, it creates an .err file:

    SolidWorks Corporation - STEP Processor Error Log
                       STEP Post Processor

STEP File Name: C:\Design-Ongoing\03 Witbox\Voron 2.4\VORON2.4_Assembly.step

Error: Encountered invalid record index.
STEP-in did not complete successfully.

mhejjas, 2022-11-07:
It was a Solidworks issue, nothing wrong with the .step file. Sorry for bothering.

closed 2022-11-07 by mhejjas

foreachthing, 2023-04-04:
@mhejjas how did you solve your problem?
I have the same issue with SolidWorks 2020 SP5.0.
Thanks.

mhejjas, 2023-04-04:
I think it was this one:
https://www.javelin-tech.com/blog/2018/11/encountered-invalid-record-index-solidworks/
or something very similar.

foreachthing, 2023-04-04:
Awesome!!! Thanks a bunch, mhejjas!

From the website, linked above:

1) Deactivate “3D Interconnect” in SOLIDWORKS Options > System Options > Import
1) Open the STEP file using text editor such as Notepad. The file will open as text.
1) Use Search/Find, and search for: `PRESENTATION_STYLE_ASSIGNMENT`
1) If this problematic data is found, replace it using the Search/Find & Replace command with: `PRES`
1) Save the file. This parameter will now be skipped during the import’s parsing process, and thus the warning will not display and the STEP should import as expected.
1) Retry the Import into SOLIDWORKS.