VoronDesign/Voron-2 #408: Multi Color 3D Print without additional filament

shahabaz, 2022-11-29:
I know this is way to far for this feature to be even considered I am just floating this idea to track some interest from others. I came across "Mimaki USA" who have developed Multicolor 3D printing successfully . Their 3D printer achieves 10M colors to be printed on the model. I guess there is an additional head to paint the surface after each layer is printed. I am not sure where to start this from may be some one can drive this feature ?

julianschill, 2022-11-29:
This is not the right place to discuss this. This is for problems and issues with the current design or manuals. For new features, mods, theory crafting and so on, join the discord or go to https://forum.vorondesign.com/ and discuss it there.

shahabaz, 2022-11-29:
Closing this issue since this is moved to forum [https://forum.vorondesign.com/threads/multi-color-3d-print-without-additional-filament.164/](here)

closed 2022-11-29 by shahabaz