VoronDesign/Voron-2 #409: V2.4 part cooling fan causing VFA's

jhenley01, 2022-12-27:
Hello everybody. I wanted to share something I've found regarding VFA's in case it helps anyone else who is having the issue. I recently built a LDO kit V2.4 and have been very happy with printer. I decided to upgrade from CW1 to CW2 and immediately started seeing VFA's on my prints. I print with black PETG a lot and being so shiny it really highlights imperfections in the print surface.

I went down the usual rabbit hole of trying to solve the VFA issue. I checked belts, tightened every screw, tuned motor currents, changed microstepping, changed from the dual gear extruder to single gear with idler bearing, checked wiring, even swapped out the LDO extruder motor for a Moons. Nothing made any difference. 

Then on one of my random test prints I noticed that the VFA issue had greatly improved. What changed? I had increased the part cooling fan from the usual 30% to 50%. I then did another test at 0% part cooling fan and it was even better!

An interesting note is that I was still using Stealthburner with my CW1 setup so there wasn't much change to the part cooling fan setup. I used the same SB parts and only added the 2 piece CW2 PCB. Since one of the PCBs sits on top of the part cooling fan there may be some vibration there now but still interesting that the artifacts lessen with higher fan speeds.

I've attached a few photos of test prints, the only change is the part cooling fan speed. I also did a test with the fan starting at 0% and increasing 10% every 5mm. For whatever reason it really doesn't like the fan running in the 30-40% range. I'll probably try replacing the fan to see if that changes anything. It's hard to get good pictures of the issue but in person the change is dramatic. 

I've seen a lot of posts about VFA's but haven't seen any that mentioned the part cooling fan. I know there are lots of things that can cause print artifacts but if you're having issues with them it may be worth looking at the part cooling fan.

[image]
[image]
[image]
[image]

nemgrea, 2022-12-27:
VFA's are most commonly caused by resonance in your printer so changing something like a fan speed can cause that resonance frequency to shift and improve or worsen the effect. changing printing speeds can also show changes in VFA's the goal is to get your setup such that the frequency of vibration of your machine do NOT match its resonance frequency and then you will not have VFA's

raymondh2, 2022-12-28:
This would do better in a discussion on our forum. https://forum.vorondesign.com/

closed 2022-12-28 by raymondh2