VoronDesign/Voron-2 #413: Y-Axis Gantry Page - 89

KingVeder, 2023-01-17:
Minor but the screw/T-Nut on the rail should start on the second one in, on both sides to allow for the Idler M3 & M5 T-Nut
[image]

nemgrea, 2023-01-18:
if the directions on page 88 are followed all of the T-nuts fit properly as shown.

[image]

newphreak, 2023-01-18:
This is very much a real issue on 300mm builds. I forget the math, but i think the 50mm difference from the 300 and 400mm rails is what makes the 300 (350mm rail) have less space here.

andreasbeh, 2023-03-28:
Can confirm this issue exists on a 300mm build, I am about 5mm short each end from being able to fit a M5 ball-spring nut. It fits just fine with a slide-in style nut, though that needs double checking for a 250mm build.
[image]

asdasdwa-v02679, 2023-05-30:
hello freinds, I hope this edit solves the problem :) let me know if it needs to be changed! 

 full page : [image]

edited parts: 
[image]