VoronDesign/Voron-2 #415: suggestion：Use the bottom of the probe to trigger the Z Endstop, not the nozzle。

fanhualei, 2023-01-31:
suggestion：Use the bottom of the probe to trigger the Z Endstop, not the nozzle。

* The nozzle is too small to trigger the Z Endstop
*  the nozzle is easy to carry excess filament ,This will trigger the Z Endstop in advance
* The Z Endstop is too close to the hot bed. When placing the PEI board, the Z Endstop will be accidentally covered.

After transformation

[image]
[image]

bdbell, 2023-01-31:
The intent of probing on the nozzle is to account for changes in Z height that occur when changing hot ends or nozzles.  To trigger on the probe body eliminates that compensation.

valadas, 2023-01-31:
With some nozzles, the nozzle might touch the bed before the probe reaches the switch. Also, I don't want to speak out of order but I think the whole idea of having a z switch may be obsolete given how good Voron TAP works :) Maybe for the next release it becomes the standard...

closed 2023-03-04 by raymondh2