VoronDesign/Voron-2 #418: Imroper use of cable chain

dkuchay, 2023-03-03:
Cable chain is designed to support high flex rated cables. Igus makes a series of cables for this application called Chainflex. Cable chain was never designed for individual conductors as Voron has allocated in the v2.4 design. That and the percentage of fill in the raceway (cable chain) is too great. Many users experience shorting around 750 hours of use due to friction within the cable chain eroding insulation. By converting to helicon or ptfe similar insulation I have noticed that these are European wiring spec and as such it drifts further from UL standards. Lower strand counts in those wires removes bend radius that is important to have in cable chain. In the TAP manual Voron made the statement that "wiring is important". If this is true would it be possible to request an update to move the print head to canbus? It would correct the mis application of cable chain being used for individual conductors. The cable chain can stay only we would need a recommendation from Igus on what shielded cable they would recommend for the amperage load at the print head. Or umbilical style wiring as in the V0. Have aurtocad electrical experience and would offer to help but I don't use social media. Chainflex link:  https://www.igus.com/chainflex/bulk-cable

julianschill, 2023-03-03:
The problem with the chainflex cables is, that with the number of wires we need, they also are not rated for the tight bending radius we have in the Vorons (around 26mm). A lot has been tried and Heluflon proofed to be quite reliable when put in the chain correctly (not under tension and fixed on the ends of the chain).

Shielded cable is not needed for CANBus, only twisted or braided pair. There are very few options available from IGUS we could use, namely CF9.05.02 and CF9.02.02 or CF9.05.04. If you want shielded, CF99.01.02 is basically the only option and it costs around 16EUR per meter.
For CAN that is feasible, but we won't fit 14 or 20 conductors in the chain with these cables.

Do you have a source that single wire shouldn't be used in cable chains? Because I am not so sure about that.

dkuchay, 2023-03-03:
Thank you for your response! I've worked in high speed robotics and Igus would repeat the industry standard of using high flex rated cables for this application of their product. Looking up Clear Automation or Mikron on YouTube will example machines I have helped to build. I was an electrical technician for both companies and spent time in Switzerland doing wiring there as well. 7 digit machinery in cost. In between projects I was taught auto cad electrical and design. I agree that too many conductors would be in a cable to support the print head as is. Thus my bus protocol recommendation. Most toolheads on robots and pick and place machines used bus protocol because reduction of wires to print head meant better efficiency and less time spent wiring along with simpler design. Not to mention less weight. How nice would it be to lighten the gantry?

The shielded portion will come in to play on the underside of the printer where more of the EMI is generated. I couldn’t wire the printer as recommended. I used wire ducting (like pandit) to make a permitter and a middle pass. With covers. Still keeps the heatsinks on the 2209’s in line with air flow. No matter the direction bus protocol signal wires will pass over EMI paths. I see many STL files for mounting their can bus board on the rear frame and I am betting this is because USB cable has shielding and they get cleaner signal by seeing the UC1 out of the electronics bay. It’s good prevantative that any signal cabling be shielded and that shielding be grounded on the source side. Will send most EMI to ground before effecting twisted pair. Do stepper motors have encoders… within industrial automation this was also another application for shielded cabling. 

As for cable chain please consider its title. It is not called individual conductor chain. If cable chain was designed for individual conductors why does Igus have a calculator on their that will forecast a high flex cable life expectancy? If Voron cares about wiring please contact Igus and ask them about their feelings on running individual conductors through a product they took the time to title cable chain. If they called it individual wire chains then you would have more of a point. Please also consider the glands they sell for each end of the cable chain and let me know if you see one that is for an individual conductor. 

I will guarantee you that they will advise a high flex cable and based on the wire count to support the print head versus the cable chain volume, they will also advise a bus protocol. You are correct, there is not enough room in the cable chain to support that thick of a high flex cable. That said it is also beyond the recommended percentage of fill as it currently sits with individual wires. You will need to inquire upon Igues as to what they recommend for fill volume on their cable chain before you need to go to a gland on each end separating the cables as fill percentage increases.  

Yes, this will nullify many of the mods that are out there. Sorry Hartk. However, it will bring your project more to standards with regards to wiring. The best option I can find is the new Big Tree Tech 2240 board with the bulk heading pins connecting the two boards. If this board does run under the rated 85 Celsius It will be good to add to stealth burner. The only concern at that point is would it make more sense to keep the cable chain in place or go umbilical. My vote is for keeping the cable chain and using a 4 conductor high flex shielded cable with the new 2240 board. Preserve vertical print volume. No disconnecting of wires when servicing via bulkhead pins and a fan can be added to the side cover more easily than the rear, where the cable chain would be mounting. 

I will happily eat my feet if Igus gives you a clean pass however I spent 5 years wiring industrial production lines for motion and this was not how it was done on any of the machines I interfaced with. Shielded high flex or bust (see problems). The European wires that are more slippery are thinner than NEC US spec guaging that UL rated. That and they are low strand cpunt causing less bend radius. I would have gotten reprimanded from my supervisor and possibly lost my job if I wired individual conductors through cable chain. Even using non high flex rated cables would have resulted in the same.

Dfdye, 2023-03-03:
There are thousands upon thousands of hours of print time that suggest that Voron's implementation of wire bundles within cable chains are not only viable, but are durable when used with high strand count PTFE insulated wires per the BOM.  So far, I have not seen any evidence that this is not the case.  Nor have we seen indications that non-shielded wire bundles suffer from interference, even when run in the electronics bay.  Other than citing guidelines that are generally applicable to much different industrial settings, I am not sure that I understand what actual problems that have been encountered that you trying to solve.  Multi-conductor cables for motion systems are often designed for industrial machinery that gets subjected to wash-downs, lubricants, particulate contamination, etc. that we do not face in a printer environment.  I frequently work with multi-conductor cable bundles that are assembled from individual strands that are encased in cable looming.  These assembled cables are subjected to some fairly extreme environments, and many are manufactured to mil-specs.  So, I want to make it clear that pre-assembled multi-conductor cables are not the only viable method for making very tough cabling for extreme environments.  

That being said, we certainly understand how cables work, and many of us even use multi-conductor cables in our printers' motion systems (myself included).  What I am curious about is why, based purely on a performance basis using empirical evidence from actual printers in the field, you believe that individual PTFE wires within a cable chain do not perform as designed.  We are absolutely willing to consider evidence-based changes to our designs, but we aren't too keen on making a change to a proven design just because it's not pre-fabricated.  For example, we use M4 thumbscrews is a manner for which they were not designed (bed supports), and they work perfectly well in that application.  We COULD change these standoffs, but there is no evidence that we SHOULD.  

I hope that this makes sense!  We are always amenable to new information, but based on the data we have from the current cable chain design, I see no new data that would suggest it should be changed.  Whatever the result, thank you for the VERY thoughtful comments!  That is always appreciated!!

dkuchay, 2023-03-04:
Thank you for your reply. Please allow me to explain better. 

First, you make perfect sense. I don’t contest anything you stated. Other than specifying cable type I mention to specific motion systems. I've worked in many industries. I see high flex cables anywhere there is motion. If you have a wash down or chemical environment then you change the outer jacketing on a motion rated cable or a high flex cable. You can’t apply that statement to all motion cables. You have a very high speed machine that perfectly falls in to the application for high flex cables and especially due to the bend radius’s imposed by cable chain. 

I apologize if I sounded imperial and thank you, I appreciate intelligent discourse. My call out is not on performance nor durability. Its on standards and application. Or better stated, manufacturers recommendations. I only mentioned performance vaguely in a lighter gantry comment. My concern on the PTFE wires are that if they are like the ones I bought from Fermio labs they will be European spec wire that is not the same as American spec wire. I commented that this drifts further from UL. Their bend radius does not lend well to cable chain. 

Yes those wires work. But it’s still not how cable chain is designed to be used and please please send inquiry to Igus asking. Its a huge no no in the any motion based machinery and just look at any examples of cable chain outside of your project until Igus returns your inquiry to see that no machine exampling cable chain shows individual wires for a quick glance. You will be hard pressed to see any expamples with anything but cables within. Not only that but high flex cables. My intent was to inform the team that cable chain is not being utilized properly and is above capacity in current design for its allowable percentage of allowable fill.  

Was taught not to impose problem without proposing a solution. I think protocol the bus makes sense as a solution to the above and to stick within manufacturer recommendations. Again, Igus to verify. Mentioned umbilical as it’s more efficient and would lighten the gantry. Or offset the weight of Tap and the 2240 board from BTT. I don’t disagree with your comments on EMI in the electronics bay either. I just harp on standards sometimes. Stepper motors are very noisy though. As I did a perimeter path with wire ducting (under desk wiring kit on amazon that was actual wire duct with cover) and it fits perfectly without blocking any fan flow, I get concerned with getting so close to noisy stepper motors with signal path. Long ago I was taught to use wire ducting when using din rail. Panel wiring standards.

Scissor lifts use cable chain. Lots of things in the drilling and rigging industries also use cable chain. CNC to name another. These will all have high flex cables within as that is the industry standard for any motion based product.

In the machines I wired we used up to 128 count multi conductor cable on the underside of the machines but we converted to high flex cables anytime a pick in place or a robot was implemented. Mikron also made machine shop tools like milling and CNC machines. All cables that support any motion in a CNC machine are flex rated. That is very close to a 3d printer. More examples where flex rated was used https://www.youtube.com/watch?v=hnWmdyBBiRk&t=2s  Yes, several levels above a Voron but a CNC machine isn’t. High flex cables should be used in all motion applications and especially with cable chain.

The 2.4 is a beautiful machine. Please don’t be offended by my comments. I chose the Voron for many positive points and when I built mine we were still ordering rolls of silicone wire and assembling our own harness. For a date I think half my parts got stuck in the Suez Canal. I prefer Samtec over Molex. Anyways The machine works and is durable as current spec is, I just know that the cable chain is being improperly used. If I was wrong would 5 can bus mods be on the market? I saw the statement on caring about wiring and that is what spawned this issue. Engineers follow manufacturers recommendations. Please inquire with Igus. Actions taken from would further the quality of the machine. 

Your project would grow by 75% or more if you handled support differently. I have to go to GitHub and read to see updates because I don’t use social media. Nor will I seek individualized support on a public chat server. I think that if it’s a site that I have to protect my kids from I shouldn’t be using it either. Your community is absolutely gigantic. Very little effort on your parts would need to be made to amass an army of agents to police and erect a traditional support venue. If linked off your main site your page rank would bump your popularity to the moon. It would also allow opportunity to centralize all the different sources for Voron info as well. I did try the chat thing. Discorse. All I can say is that must work much better for you than it did for me. That is a place to get scammed and prowled. I have to think that most above a certain age will do a 180 once they see the support venues offered. This is a different issue and not related to the named issue of improper use of cable chain though. Just 2 cents.

raymondh2, 2023-03-04:
Until we find a suitable cable that works with the chain sizes we currently use the recommended high strand count PTFE will remain the standard. I am sure that we can get many more hours out of Igus Chainflex but at the end of the day this is an open source DIY hobbyist 3D printer. If you need something that lasts longer feel free to change the parts to your needs.

closed 2023-03-04 by raymondh2

dkuchay, 2023-03-04:
Acknowledged. So you will offer verdict without checking with the manufacturer? Sent from my iPhone

stvptrsn, 2023-03-04:
There is no need to check.  We know that IGUS does not have chainflex that will meet our spec configuration needs.   Everything else I would comment here is contained in Dave's reply above.

julianschill, 2023-03-04:
On the IGUS website is a product finder. If you filter for our bending radius the only wires that remain are the CF9 wires I mentioned earlier. We want to avoid single source parts on the BOM, though. So the only options I see are:
1. Keep it as it is with single conductors in the chain
2. Use Umbilical with high conductor count
3. Use CAN either in the chain or as umbilical

Problem with CAN is that many users struggle setting it up, flasing and updating the boards and so on and we experience lots of problems with klipper errors (timeouts, communicatione errors, "Timer too close" errors. I think it is not stable enough to use it as the default. Umbilical with many wires is doable, but I prefer the chains here.

PS: If you don't want to use Discord, consider using the new forum at https://forum.vorondesign.com
I think that would be better suited for such discussions.

dkuchay, 2023-03-04:
Thank you and I understand user setup concerns. Hopefully your new forum can centralize knowledge. Lower operating temp can boards may address the errors. Just look at airflow needed on a 2209. Also why I mentioned standards in signal cabling. I will articulate more in the forum, thank you for mentioning. Just to review, when it is stated that Igus does not have a product that meets our spec needs this aligns with my using iOS going against manufacturer recommendations as well as improper use of cable chain. There is not a drag chain out there that will meet the current spec. That was the purpose of posting the issue. This is not solved but if you wish to mark it complete it demonstrates improper use of GitHub and poor issue tracking. Technically this issue should stay open lending way to v3. V2.5? Not for me to decide. By marking something that is not solved as solved you make an interesting statement.  I have read some very toxic things regarding this topic. And they just came.back to memory. If voron team refuses to acknowledge and would rather make legitimate mid usage issues go away rather than to address that is shame on you. Progress looks better than preserving image in the end users eyes. Sent from my iPhone

dkuchay, 2023-03-04:
Consideration:The cable chain servicing the z axis could grow in height with its only limitation being either its own bend radius or a particular mount point for a clicky probe. From there put a stand off on the gantry’s rear cross beam. If one piece of cable chain that was rotated 90 degrees ,sweeping both x and y axis, connected from this stand off to the print heads mount point for the cable chain this too could grow in height thanks to the afforded room behind the build plate. It would not be as pretty but it may place within manufacturers recommendations on percentage of fill and a multi conductor flex rated cable may have enough room with added height to the cable chains. Igus sells to individuals as well as automationdirect.com and if you need a third you could use digikey.  If your project is closed source and I have misunderstood its nature by being on an open source venue I do apologize for misunderstanding.

Dfdye, 2023-03-04:
> Technically this issue should stay open lending way to v3. 

I would disagree.  As Steve and Ray have both stated, we have considered this recommendation and have decided, based on empirical evidence and informed engineering design choices, to close the issue and retain the current design since it has proven to work effectively if the BOM recommended materials are used.  If you do not agree with this decision, that does not mean that it was either toxic or that it was incorrect.  Nor are we attempting to shame anyone.  The decision means that the engineering team has considered the question and made a choice based on the reasons stated above.  We completely understand that you disagree with this decision, and that is your right.  But having your recommendation be considered and then rejected based on logical arguments grounded in empirical evidence that have been developed in the context of overarching design goals is not a sign of toxicity or "shame" within  the Voron community, but rather a sign of a healthy decision making process that has been explained openly when questioned.

Dfdye, 2023-03-05:
To add a bit of information, spoke this afternoon with some of our engineering folks who originally made the design decision with regards to cable chains.  They DID talk to Igus regarding suitability of their (Igus's) products for our application, and they (Voron designers) came away from that meeting with lots of information, but no product that Igus made that would suit our needs with regards to cabling.  I, personally, was not involved in those discussions, but I did want to relay this info since OP raised the question regarding why we would close the issue before talking with Igus.  The answer, quite simply, is that we already have.  The discussions were focused on trying to identify products that they sold that would suit our needs, and not specifically focused on technical qualities of cable chains.  

Even if we had not, we have empirical evidence that our solution is robust.  Our experimental testing during development along with many, many thousands of hours of field use has shown us that wear/failure rates for our application are acceptable.  This abundance of evidence combined with the fact that we have extensively discussed cables and wiring in the past is why we were able to quickly close this issue, even if we had never spoken with Igus.  

I hope that this explanation helps to better explain why our decision making process was so quick with regards to this particular issue.

dkuchay, 2023-03-06:
> Thank you for checking with Igus and for the added context. It does not change the fact that American manufacturers of cable chain consider individual conductors as outside of what they would recommend. True Engineers consider manufacturers white papers to be strict adherence. This is something the NEC regulates in America. Like wire nuts there is no standard on how many turns it takes best it’s assumed that best practice will be consisdered by those using. Current implementation is not utilizing best practice and is in my opinion, in need of uipdating, Acknowledge you have 1000’s of hours but you are not wiring each Voron. This exchange and decision to knowingly go against mgr recommendations on flexible raceway products has caused me to donate my Voron to a worthy cause and I will be moving to a project based printer more observant to standards. This issue should not be closed until you figure out how to use cable chain within manufacturers recommendations as you are still not applying it correctly. Title of the issue being key. Looking at the VZbot braces helping input shapter I have to think that 40 series extrusion is the next step. That would be logical for your use of cable chain too. 20 series not holding inertia too well. It’s amazing to me the range of responses I heard up to someone checking with Igus. Again, thank you for checking.

Originally I thought that the canbus errors you are reporting, although more and more of the community is adopting, was due to some outlying factors. After looking in to this more I am seeing BTT as a company that is using your user base as gigantic beta testing field. They have not implemented several things correctly and using less of a Swiss army knife may also be a forward looking consideration.