VoronDesign/Voron-2 #421: BOM should include 24V & 5V PSU Terminal Covers

dekay, 2023-03-18:
The BOM has a "Mean Well LRS-200-24 PSU" and a "Mean Well RS-25-5 PSU" but it should also include the transparent terminal covers as well given that this is a safety issue and they costs pennies. The Mean Well cover is also better than a printed version because it is transparent, allowing you to see the screw terminals.

The appropriate cover for the 24V PSU should be named "Mean Well TBC-09 Terminal Cover" and [it is available at Digikey](https://www.digikey.ca/en/products/detail/mean-well-usa-inc/TBC-09/7707110).

The appropriate cover for the 5V PSU should be named "Mean Well TBC-05 Terminal Cover" and [it is also available at Digikey](https://www.digikey.ca/en/products/detail/mean-well-usa-inc/TBC-05/7707107).

Note that the Trident BOM has one for the 24V PSU but not the 5V one.  I logged that as a separate issue [here](https://github.com/VoronDesign/Voron-Trident/issues/104).

Tenkir, 2023-09-17:
Genuine Mean Well PSUs include these covers in the box. Or at least, every one I've ordered from Digikey or Mouser includes it. Questionable vendors from Aliexpress and Amazon sources often do not, but the BOM (IMO) should not encourage people to source critical electronics components from unrepeatable sources.

nemgrea, 2023-09-17:
Real meanwell psus do not always come with terminal covers. They have their own part number. The lack of a terminal cover does not indicate a counterfeit product.

The bom encourages a specific component from meanwell. The sourcing guide provides likes to parts that we have actually purchased and can verify. They are not random links..

dekay, 2023-09-18:
@Tenkir The genuine Mean Well supply I ordered from Mouser did not include a cover and that is why I raised this issue in the first place.  I agree with @nemgrea.  The BOM is simply not consistent and it makes no sense to spec a cover for one and not the other.  You _can_ print one after the fact, but given that this is a safety issue, that it isn't in the printed parts list, that it might be somebody's first printer and for a while they'd be running with exposed AC terminals, and that the cover itself costs pennies, the BOM should list both.