VoronDesign/Voron-2 #423: XY Joint Bearing Stack Misalignment Relative to Belt Path

stevereno30, 2023-03-29:
The V2 XY joints position bearing stack idlers 0.3mm out of alignment with the belt path in the CAD. 
[image]

This image shows the bearing stack shifted 0.3mm up.
[image]

raymondh2, 2023-03-29:
The belt in the CAD is just a reference. Not an accurate depiction

stevereno30, 2023-03-29:
> The belt in the CAD is just a reference. Not an accurate depiction

Unless it is designed with a slope relative to the XY plane, shouldn't it be perfectly aligned with all fully constrained bearing stacks?

stevereno30, 2023-03-31:
Upon further investigation, it looks like the issue has to do with the X extrusion positioning. It is 0.3mm out of plane with the rest of the gantry. The image below shows a sketch line that sits on the top of both the x extrusion and the y extrusion, but are aligned relative to the x axis. Notice how the white line pierces the y extrusion.
[image]

nemgrea, 2023-04-03:
the entire X axis gantry is just shifted down by 0.29999268mm
[image]

xhxhwkl, 2025-02-20:
By measuring the Y-axis gauge to the height of the aluminum profile, it can be seen that it is 10.3mm. The entire x-part can be moved up by 0.3 to the correct position, and it will be found that the corresponding bearing position matches the belt