VoronDesign/Voron-2 #431: Problem with bottom panel

pOmelchenko, 2023-05-13:
Hello! I found the bug with bottom panel.

[image]

I can't open that panel because foots doesn't give enough space. In documentation, we can see same foots.

[image]

and here I'm try to close panel

[image]

[image]

I was download latest cad file and checked how it works — it doesn't work

[image]

bombela, 2023-05-20:
I have the same issue, but I can pull a bit on the panel as I open to clear the feet for what is worth.

timmit99, 2023-07-06:
Let me take a look at the cad and see if I can come up with an updated trim guide for the bottom.

timmit99, 2023-07-06:
[image]

So i was able to shift that cutout ≈5mm and that give's enough clearance. Let me work on a new bottom panel drawing and trim tool

pOmelchenko, 2023-07-06:
Good solution, but now this corner needs in new feet plastic parts for final construction