VoronDesign/Voron-2 #439: Manual: Z Idler

1John419, 2023-07-13:
On page 48, the GT2 20 Tooth Idler is shown fixed by a M5x30 BHCS screw.

Although not explicitly stated, I had assumed that it was screwed directly into the plastic of the Z Tensioner. However, printed part has no retention on the screw and the STL clearly shows a through 5.0mm diameter hole. As I measure a typical M5 at 4.75mm diameter, this appears to be an intention slip fit.

My only guess is that the design intends to utilize belt tension to retain the screw. If that is true, as a first build, this is a very confusing omission of information.

I have suspended my build until there is clarification of design intent.

ensoniq2k, 2023-07-14:
It's screwed directly into plastic. In my case I didn't even notice because of warping of the ASA the hole needed to be drilled out anyway.

nemgrea, 2023-07-14:
> As I measure a typical M5 at 4.75mm

4.75mm is out of din spec for the major diameter of an m5 bolt.

1John419, 2023-07-14:
Yes, I have confirmed that some of the Formbot kit fasteners are out of spec. I have some other M5 fasteners that are in spec (4.86mm diameter) which do produce a very slight bite when threaded. I'll order some new fasteners (confirming their spec), reprint the parts, and see if that yields the desired results.

As a first time builder, I would comment that the manual could be clearer regarding this matter. Other threaded directly into plastic parts (e.g., A/B Drive and XY Joint) have printed in place threads. For the Z Idlers, perhaps making these mirrored parts with printed in place threads on the outward arms (and slightly oversized hole on the inward arms) would provide a better solution.

1John419, 2023-07-17:
OK. Got some in spec fasteners (4.96mm). Reprinted parts. Fastener bites into part allowing a slight torqueing and solid retention. Build restarted.

Opened a ticket with Formbot to report the issue with fasteners.

closed 2023-07-17 by 1John419