VoronDesign/Voron-2 #456: What is estimated cost of a printer?

skorphil, 2023-11-14:
Hi, wahat is approximate cost of this build?

ensoniq2k, 2023-11-14:
Depends on the size you're choosing. You can check out the LDO, Formbot and FYSETC kits. They include all the parts necessary. If you're going to source yourself nobody can tell you how much everything costs where you live.

skorphil, 2023-11-14:
Thanks, i need 100x100x100 print area. But as far as i understand, size not make much difference. Anyway similar electronics - most of the cost

ensoniq2k, 2023-11-14:
In that case you're probably better off with a Voron 0.2. The parts of the Voron 2 are a bit big for that build size I guess. Voron 0.2 is really fast.

skorphil, 2023-11-14:
> In that case you're probably better off with a Voron 0.2. The parts of the Voron 2 are a bit big for that build size I guess. Voron 0.2 is really fast.

Thank you, will have a look

skorphil, 2023-11-15:
> In that case you're probably better off with a Voron 0.2. The parts of the Voron 2 are a bit big for that build size I guess. Voron 0.2 is really fast.

looks nice. But bigger than actually needed. It is also using direct drive as 2.4. I think bowden rules

nemgrea, 2023-11-22:
this is not an issue, this is a question and should be posted on the forum/discord/FB group instead 

closing for that reason only.