VoronDesign/Voron-2 #457: BOM list export

tristelune, 2023-12-24:
Hello

On the main Voron web site, there is an issue with the exported BOM list of the Voron 2, due to the fact the csv file is comma separated, and in some lines there is text with comma like this one (extract from the csv file directly):
Cables,Nylon Cable Ties, Small (.07-.10", 1.8-2.7mm wide),42,

This text should be in one cell :
Small (.07-.10", 1.8-2.7mm wide)

but it is split in two cell due to the comma inside it, messing with the table when imported in open office, and probably also in excel.

To solve it, all the BOM lists should be exported using semicolon, or forbid the usage of comma in cell.

Thanks

nemgrea, 2024-02-29:
fixed

closed 2024-02-29 by nemgrea

tristelune, 2024-03-01:
Hello

Just tested to import the bom list after the fix and it's work perfectly now.

Thanks for change