VoronDesign/Voron-2 #459: Defualt Min Extruder Temp is Problematic

sirbrialliance, 2024-01-06:
In Voron2_Octopus_Config.cfg, and perhaps others, the default extruder min_temp is 10° C. If you store your printer in a cool, but not freezing, location, this can trip you up when the printer suddenly stops working in the winter.

Real-life issue I had (my config is customized, but the min_temp default is unchanged):
[image]

julianschill, 2024-01-07:
I don't think that's a problem. Most printers don't experience such low temperatures and for those who do the minimum temperature can easily be changed. 
The minimum temperature is a security feature to catch bad thermistors or wiring so I wouldn't change it in the default config.

sirbrialliance, 2024-01-10:
That's an understandable position.
The tricky thing is that it's not likely to come up while you're setting up the printer. Perhaps a comment pointing out that this default may not be a good value if you store your printer in a cold location would be the best way to address this?

closed 2024-02-29 by nemgrea