VoronDesign/Voron-2 #475: Manual Page 33 "M5 Shim" can be ambiguous

Altirix, 2024-04-20:
This should be explicit and state "M5 Shim 1mm" as there are two types of M5 Shim in the fysetc fastener pack for V2.4 (1mm, 1.5mm)

[image]

the CAD must be opened to confirm what shim this references.

[image]

Altirix, 2024-04-20:
seems these 1.5mm shims are only included by fysetcs fastener kit and are not part of the official BOM.

ill keep this up as i think it does still raise a genuine question on the ambiguity when many vorons are built using these kits

nemgrea, 2024-05-14:
what fystech chooses to include or not include in their kits is their responsibility. our hardware page at the beginning of the manual adequately covers the m5 shim definition and where they can be found, also 1.5mm shims are not part of the voron bom so their inclusion in the fysetc kit cannot be controlled or managed by us.

[image]

closed 2024-05-14 by nemgrea