VoronDesign/Voron-2 #478: I want to edit some parts, what software should I use? The files in the CAD folder cannot be edited when opened.

jackshi618, 2024-07-14:
I want to edit some parts, what software should I use? The files in the CAD folder cannot be edited when opened.

hartk1213, 2024-07-14:
@jackshi618 what software are you using to open them?

catalinalb, 2024-07-31:
Good day, I am tring to open the STEP file in Solidworks Professional 2020 and in Solidworks Maker 2024. 
No succes, I have plenty of RAM and latest i7....
Could someone check this? 

Thank you

wolfgangmauer, 2024-07-31:
Opening the STEP file with FreeCAD doesn't cause any problems

TheLexoPlexx, 2024-08-03:
> Good day, I am tring to open the STEP file in Solidworks Professional 2020 and in Solidworks Maker 2024. No succes, I have plenty of RAM and latest i7.... Could someone check this?
> 
> Thank you

No issues in Solidworks 2024 and I've used 2020 before.

catalinalb, 2024-08-03:
I had to open the STEP file in FreeCAD, export it as a new STEP file. Only then, Solidworks was able to open it.

closed 2024-08-05 by nemgrea