VoronDesign/Voron-2 #490: Low importance  - Broken Link- Manual -  page 181 - PSU VOLTAGE CHECK

bom-sourcingguide, 2024-12-17:
(Low importance)
The link "Refer to the Mean Well LRS-200 datasheet for possible settings (voron.link/e0szdyh)."
is broken
goes to 
https://raw.githubusercontent.com/VoronDesign/Voron-2/Voron2.4/Manual/voron.link/e0szdyh

404: Not Found

nemgrea, 2024-12-18:
this has been fixed, the link should now redirect to the correct spot

closed 2024-12-18 by nemgrea

bom-sourcingguide, 2024-12-24:
Hi @nemgrea Thanks for fixing the link, the URL works, just the clickable version of it is broken going to file:// (probably missing like https:// prefix or something)

Also doing some QA these two are not clickable: (all other links are good)
Page 160 - voron.link/3bwwnqy
Page 207 - voron.link/n9i7lss